LilyGO/TTGO-T-Beam #4: Issue running GPS code

woody121, 2018-08-12:
Super handy library, the LoRa stuff works great.
Trying to read off the GPS, I am getting:

Archiving built core (caching) in: /var/folders/fs/yrrtd9hn7p1bsz8n_57_69cc0000gn/T/arduino_cache_893081/core/core_esp32_esp32_ttgo-lora32-v1_FlashFreq_80,UploadSpeed_921600,DebugLevel_none_279fa844f13a7dc74d194a5db99730eb.a
arduino.ar(HardwareSerial.cpp.o):(.bss.Serial1+0x0): multiple definition of `Serial1'
sketch/GPS.ino.cpp.o:(.bss.Serial1+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board TTGO LoRa32-OLED V1.

I added the TinyGPS++ library to my Arduino IDE to get to this point. I'm a long time mobile dev doing my first dabbles into Arduino stuff so forgive me if this is a basic thing! Thanks

woody121, 2018-08-12:
If I comment out the line "//HardwareSerial Serial1(1);" it runs, but I get no data from the GPS (all 0s).

jcwren, 2018-08-12:
Try reversing the TXD and RXD pin numbers. My code was using the Espressif esp-idf platform instead of Arduino, but the pins numbers listed are backwards.

And if you're lucky, you won't have a counterfeit uBlox NEO-6M like my two do. It'll work as a GPS, but only with basic NMEA strings. You won't be able to use uBlox commands, or the NMEA PUBX commands.

woody121, 2018-08-12:
Thanks @jcwren for the quick reply, sadly no luck. Any suggestions about further steps to troubleshoot? 

I should note that I have the TTGO board without a OLED (unlike as indicated in my initial post), specifically: https://de.aliexpress.com/item/TTGO-T-Beam-ESP32-433-868-915Mhz-WiFi-wireless-Bluetooth-Module-ESP-32-GPS-NEO-6M/32875743018.html

jcwren, 2018-08-12:
Sorry, no other ideas. Once I got the TXD and RXD straightened out, it started spitting out NMEA data at 9600 baud.  The GPS is definitely on UART1.  That's the same one I have, only through Banggood.

woody121, 2018-08-12:
Solved! Being in the middle of an old building was the problem, once I was up on the roof I was good to go. 

For future reference, the key was noting that there are two status LEDs next to the NEO-6M chip, the blue one for GPS power and the red for GPS lock. This red LED should be blinking - if not, then no GPS fix.

closed 2018-08-12 by woody121

KommandorKeen, 2018-11-07:
Even simpler for diagnostic purposes just load a serial echo/passthrough and check out what is coming out.
I had to comment out the Hardware Serial line too.
But found pin 12 (as was written on the board) was the correct pin to receive info.

With serial passthrough I could use the unit as a gps input to QGIS.

Xerxes3rd, 2019-05-14:
For anyone interested, here's a sample sketch that's basically a pass-through to the GPS.  u-blox u-center works fine with it, and is able to enable/disable sentences, etc.

```
// Note these are opposite from the silk screen on the board
#define GPS_RX 12
#define GPS_TX 15

void setup() {
  Serial.begin(9600);
  Serial1.begin(9600, SERIAL_8N1, GPS_RX, GPS_TX);
}

void loop() {
  while (Serial1.available()) {
    Serial.write(Serial1.read());
  }
  while (Serial.available()) {
    Serial1.write(Serial.read());
  }
}
```

lopoty, 2022-02-26:
My ttgo t-beam continues to has problems as the code cannot be compiled. What to do?
The error is 
Archiving built core (caching) in: /var/folders/fs/yrrtd9hn7p1bsz8n_57_69cc0000gn/T/arduino_cache_893081/core/core_esp32_esp32_ttgo-lora32-v1_FlashFreq_80,UploadSpeed_921600,DebugLevel_none_279fa844f13a7dc74d194a5db99730eb.a
arduino.ar(HardwareSerial.cpp.o):(.bss.Serial1+0x0): multiple definition of `Serial1'
sketch/GPS.ino.cpp.o:(.bss.Serial1+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board TTGO T-Beam

If I put a comment in the code Serial1.begin(115200, SERIAL_8N1, 12, 15);   //17-TX 18-RX 
it doesnot have problems.