LilyGO/TTGO-T-Beam #5: Which board I have to select?

covra, 2018-08-24:
In the Arduino IDE, which board I have to select to upload those sketches?

thx

covra, 2018-08-24:
In addition.. the sketch doesnt compiles:

sketch\GPS.ino.cpp.o:(.bss.Serial1+0x0): first defined here

uart gpio problem????

mhjxs4all, 2018-08-27:
Covra, see new pull request, serial1 is a restricted label, change al Serial1 labels to GPSSerial1 the it works.

covra, 2018-08-29:
Thank you, now it works, but what I really was looking for, was a full loraWAN compliance sketch

mhjxs4all, 2018-08-29:
You're welcome,  It was a quick fix. 
I was looking for the schematics of the t-beam with the silk screen t22_v06 
In my search I found the deux /ttn-mapper source that could be a good base to start from on github

covra, 2018-08-29:
very thankful! The code seems to work perfect! 
... Now my headache is the 1-channel-single-gateway  ;)

cnorman27, 2019-07-26:
If you have them imported, the ESP32 Dev Board in Arduino will also work.

SamiurRahman1, 2020-01-19:
> In the Arduino IDE, which board I have to select to upload those sketches?
> 
> thx

Hello. i am struggling with this problem as well. Could you please tell me which board you selected finally?
thanks

rhinotek-NL, 2020-05-16:
i am facing upload problem , TTGO Tbeam says "fetal error timeout" at the time to uploading any sketch