LilyGO/TTGO-T-Beam #6: Where are the libraries needed to get this code to work?

jeprince, 2018-08-24:
There are several different versions of libraries online, which ones are needed for this code to work?

jeprince, 2018-08-24:
I can't get anything to work. I can't find documentation on anything. Not even sure what board to select in the Arduino IDE to upload code to....

covra, 2018-08-29:
- Preferences > URL Additional boards `https://dl.espressif.com/dl/package_esp32_index.json`
- Board: `ESP Dev Module, DIO, 4MB(32MB), 80 MHz, 115200`

This work for me ;)

jeprince, 2018-08-30:
That didn't work for me, i had to choose "Microduino-CoreESP32", also had to use the driver (i'll link below) for Windows10 to even be able to communicate with it. Now that I can get the GPS.ino code to work.....I'm trying to get the LoRa_Sender code on here to work but i cannot find the right libraries SPI.h, LoRa.h, Wire.h. There are several different versions online and i have not stumbled upon the right ones

https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers