LilyGO/TTGO-T-Beam #8: Neo6 RF_IN wiring is incorrect.

lumpylumpy, 2018-08-29:
Hardware: T-Beam T22_V06 20180706
Firmware: SoftRF 1.0-rc4+
[image]
[image]

R33 is shown as a 22uh choke going between the center pin of the ipx connector and RF_IN. A choke in this position would not be useful in passing the 1575 mhz satellite signals from the patch antenna to the RF input of the Ublox Neo6 gps. A capacitor could work to block the dc to the antenna input but in my board this item measured .6 ohms and when taken off measured 17uh on my meter, so it was in fact an inductor
[image]
. The normal Ublox wiring has a choke in the line from the antenna connector pin through a limiting resistor to the VCC_RF that powers the patch antenna. The existing wiring does work, but is improved by removing the choke and putting it in series with the 22 ohm resistor R28. The wiring from the RF_IN was then jumpted to the center pin of the ipx connector .  In my case this improved the number of satellites tracked by 1 or 2 and improved the position error. I suggest this change me made to the pcb before the next production run.

David VE3KCL

jcwren, 2018-08-29:
If you're feeling adventurous, pop the cap off the NEO-6M and see if you have an AT6558.  The cap is soldered so poorly that you don't need to heat it. Both of mine have counterfeit uBlox modules.

lumpylumpy, 2018-08-29:
Hi JC  nothing wrong with an AT6558... much faster than a Neo6  I have tested them to 48000m with no high altitude string required ... not a downgrade from a Neo6 but deaf to ubx..  a little bit of slight of hand with the label but the T-Beam works well with SOFT-RF and this  gps chip. I have not pried my lid off but this thing locks so much faster than a Neo6 ever did, I would not be surprised....   Dave

jcwren, 2018-08-29:
The MAJOR problem is they're selling as a NEO-6M (not necessarily TTGO, but
whomever they buy parts from). And if you have an application that uses UBX
(I do), and you're expecting the GPS to speak UBX (and I do), then when it
doesn't, it's a real problem.

--jc

Save a life; adopt a shelter animal.

tedder, 2019-08-25:
Yeah, note the datasheet != the protocol. I couldn't find an english version of the protocol for the AT6558. It's fine if you can handle the default strings.