LilyGO/TTGO-T-Beam #9: SPI Pins

jdbriggs74, 2018-09-01:
Hi
Please can you advice where the SPI pins are.

On the PCB it has printed SCK=io5, MOSI=io27, MISO=io19

How do I physically connect to these pins? I can't see any pin outs on the board with these io numbers.

Thanks in advance for any help.

Johnathan

lemmingDev, 2018-10-03:
There are 2 buses we can access

IOMUX pins for SPI controllers are as below:

Pin Name	HSPI	VSPI
                        GPIO Number
CS0*	           15	  5
SCLK	           14	18
MISO	           12	19
MOSI	           13	23
QUADWP	      2	22
QUADHD	      4	21

note * Only the first device attaching to the bus can use CS0 pin.

https://github.com/espressif/esp-idf/blob/master/docs/en/api-reference/peripherals/spi_master.rst

Don't have my t-beam yet, but will try an SPI display when I get a chance.

lemmingDev, 2018-10-03:
Seems there are 2 buses we can access

IOMUX pins for SPI controllers are as below:

Pin Name	HSPI	VSPI
                        GPIO Number
CS0*	           15	  5
SCLK	           14	18
MISO	           12	19
MOSI	           13	23
QUADWP	      2	22
QUADHD	      4	21

note * Only the first device attaching to the bus can use CS0 pin.

https://github.com/espressif/esp-idf/blob/master/docs/en/api-reference/peripherals/spi_master.rst

Only problem is that a lot if the required pins seem to be used for connecting the LoRa and GPS.

They mention an IO multiplexer. Maybe we can realocate the pins.

Don't have my t-beam yet, but will try an SPI display when I get a chance.

kotakomputer, 2019-05-11:
<table>
<thead valign="bottom">
<tr><th rowspan="2">Pin Name</th>
<th>HSPI</th>
<th>VSPI</th>
</tr>
<tr><th colspan="2">GPIO Number</th>
</tr>
</thead>
<tbody valign="top">
<tr><td>CS0*</td>
<td>15</td>
<td>5</td>
</tr>
<tr><td>SCLK</td>
<td>14</td>
<td>18</td>
</tr>
<tr><td>MISO</td>
<td>12</td>
<td>19</td>
</tr>
<tr><td>MOSI</td>
<td>13</td>
<td>23</td>
</tr>
<tr><td>QUADWP</td>
<td>2</td>
<td>22</td>
</tr>
<tr><td>QUADHD</td>
<td>4</td>
<td>21</td>
</tr>
</tbody>
</table>
Unfortunately the TTGO PIN mapping not same as ESP32 documentation, i.e:
MISO 12 and 19 both not exposed (TTGO uses 19 for MISO but PIN 12 not exposed). Any information if PIN 12 used by GPS?

theblujuice, 2019-05-19:
> Seems there are 2 buses we can access
>
> IOMUX pins for SPI controllers are as below:
>
> Pin Name HSPI VSPI
> GPIO Number
> CS0* 15 5
> SCLK 14 18
> MISO 12 19
> MOSI 13 23
> QUADWP 2 22
> QUADHD 4 21
>
> note * Only [...]

Heya lemmingDev
Did you get your t-beam yet and get an So I display working yet? I would like to order one but want to ensure a SPI display works first.

lemmingDev, 2019-05-20:
I was able to connect an SPI display to the T-Beam and get the NES emulator working on it.

The SPI pins can indeed be changed...

FLYINGSC0T, 2019-06-12:
Hi lemmingDev,

Would love to know which pins you used to get an SPI display working on a T-Beam.

FrancisObicuelo, 2019-10-08:
rjiretoiehntkwert

FrancisObicuelo, 2019-10-08:
Hi @lemmingDev  , I would like to know how did you connect your T-BEAM with SPI display.

I've been searching and I still can't remap MISO and CS pins...

I'm looking foward to hear from you,
Francisco.

FLYINGSC0T, 2019-10-09:
Hi FrancisObicuelo, 
For information, if you are using Bodmer's TFT_eSPI library you can now re-map the display to use the HSPI bus on the T-Beam. If this is your case, you do need to change some defines in the setup files for it to work.

FrancisObicuelo, 2019-10-09:
Hey @FLYINGSCOT,

Thanks for your reply!
Yes I’m using Bodmer’s tft_eSPI library and i can’t move on... could you please send your code / exemple? How did you do that?
And if you re-map those pins, can you still use Lora?

im looking foward to hear from you.
Francisco

FLYINGSC0T, 2019-10-09:
Ok Francisco - I’ll get back to you tomorrow as I’m waiting to board my flight.

FLYINGSC0T, 2019-10-10:
Hi Francisco,

Firstly, on my T-Beams the pins are mapped as follows:

For LoRa:
RST 23
NSS/CS 18
SCK 5
MOSI 27
MISO 19

For the GPS:
TX 12
RX 15

As far as I'm aware, the HSPI pins on the T-Beam are:
HSPI CLK 14
HSPID (MOSI) 13
HSPIQ (MISO) 12
HSPI CS0 15

So, unfortunately as the GPS uses GPIO 12 for TX and 15 for RX we cannot use these pins to connect to the MISO pin or CS on the TFT. Therefore, I suggest connecting TFT MISO to GPIO 4, TFT CS to GPIO 2 on the T-Beam. Connect the MOSI on the TFT to GPIO 13 and the TFT CLK to GPIO 14. 

You now need to setup the TFT_eSPI library:
In the User_Setup.h file find the line #define USE_HSPI_PORT and uncomment it.
/ The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
#define USE_HSPI_PORT

Next, you need to define the type of display you are using e.g. ILI9341, ILI9488 etc.
In the User_Setup_Select.h file find the section similar to the following:
//#include <User_Setups/Setup1_ILI9341.h>  // Setup file configured for my ILI9341
//#include <User_Setups/Setup2_ST7735.h>   // Setup file configured for my ST7735
//#include <User_Setups/Setup3_ILI9163.h>  // Setup file configured for my ILI9163
//#include <User_Setups/Setup4_S6D02A1.h>  // Setup file configured for my S6D02A1
//#include <User_Setups/Setup5_RPi_ILI9486.h>        // Setup file configured for my stock RPi TFT
//#include <User_Setups/Setup6_RPi_Wr_ILI9486.h>     // Setup file configured for my modified RPi TFT
//#include <User_Setups/Setup7_ST7735_128x128.h>     // Setup file configured for my ST7735 128x128 display
//#include <User_Setups/Setup8_ILI9163_128x128.h>    // Setup file configured for my ILI9163 128x128 display
//#include <User_Setups/Setup9_ST7735_Overlap.h>     // Setup file configured for my ST7735
//#include <User_Setups/Setup10_RPi_touch_ILI9486.h> // Setup file configured for ESP8266 and RPi TFT with touch
You need to uncomment only the one file that you want to use. In that file, let's say it's for an ILI9341 display you need to edit the parameters for the connections e.g.:
// HSPI
#define TFT_SCLK 14
#define TFT_MISO 4 // ILI9488 - connect T_D0 but leave TFT MISO disconnected if other SPI devices share MISO
#define TFT_MOSI 13
#define TFT_CS   2
#define TFT_DC    32 // perhaps as these are io pins ?
#define TFT_RST  33 // perhaps ?
Finally, you need to ensure the define is enabled for your particular display in the TFT setup file:
// Only define one driver, the other ones must be commented out
#define ILI9341_DRIVER
//#define ILI9488_DRIVER
//#define ST7735_DRIVER
//#define ILI9163_DRIVER
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
Connect everything, compile and run. Fingers crossed it should work for you.
George

FLYINGSC0T, 2019-10-10:
For information, I've got LoRa, GPS, Barometer, ILI9488 display, led and sound alarms all being controlled by multi-threading on an ESP32 (not T-Beam however). I'm just waiting for the delivery of a CO sensor to incorporate into the project.

Evilgeniusnerd, 2021-02-26:
@FLYINGSC0T Do these pins you said above match up to the TTGo T-Beam V1? I tried to follow these but setting my defines in my INO trying to use an SD Card device and I run into issues.

FLYINGSC0T, 2021-02-26:
Hi Evilgeniusnerd . The pins I mention match up to a 38-pin ESP32 Dev Module v0.7 The pins on v1.0/v1.1 modules are slightly different e.g. the TX on the GPS is GPIO 34. 
See the pinouts here:
[image]

It's a pity that the GPIO's for LoRa are not broken out on T-Beam's as the SPI bus using these pins the cannot be shared with other devices and therefore very few pins are available for other uses.

Evilgeniusnerd, 2021-02-26:
@FLYINGSC0T  Yeah I am having trouble trying to figure out what pins I can use. Yeah it really is annoying they didn't bring those gpios out because that was my first idea and then was like well darn

lemmingDev, 2021-02-27:
Can't almost any pin can be used for SPI on the ESP32?

lemmingDev, 2021-02-27:
https://rntlab.com/question/esp32-spi-port-switching/

FLYINGSC0T, 2021-03-02:
Hi all. Following on from our previous discussion, I've been able to get my T-Beam v0.7 working with an SPI display, LoRa, GPS and a BMP280 barometer. However, I haven't been able to get it to work with Bodmer's TFT_eSPI library (LoRa doesn't work !!) but it does work with Adafruits IL9341 library and may help you get started. I hadn't actually used an SD card reader previously so it was also a bit of a challange for me !! :) I used a different SD library (link below) because with the Arduino SD library the SD.begin() function cannot be overloaded with the CS, MOSI, MISO and SCLK pins that you want to use. You can only pass the CS and it uses the default pins which are used by the LoRa module on the T-Beam.

Here's my setup:

The T-Beam I'm using is the v0.7 but should also work on the v1.0 / v1.1

The IL9341 display I used can be found here:
https://www.aliexpress.com/item/32960934541.htmlspm=a2g0s.9042311.0.0.27424c4d11DB7z

The SD card library can be found here:
https://github.com/nhatuan84/esp32-micro-sdcard

My configuration is as follows:

#include <SPI.h>
#include <LoRa.h>
#include <Wire.h>
#include <SoftwareSerial.h>
#include <mySD.h>

#include "Adafruit_GFX.h"
#include "Adafruit_ILI9341.h"
#include <M6NTinyGPS++.h> // Renamed so to use a NEO M6N GPS - There's also another library available modified by Andrea Spiess for the NEO M8N GPS
#include <BMP280.h>

// Define the chip select GPIO for the SD Card - T-Beam
#define SD_cs 25 // goes to the CS pin on the SD card reader

// Define an instance for the file for the SD card
File myFile;

//  Define the LoRa module GPIO pins - these are the default GPIO's that are actually connected to LoRa module on the T-Beam
#define LoRa_SCK      5     // GPIO5  -- SX1278's SCK          
#define LoRa_MISO    19     // GPIO19 -- SX1278's MISO - may not be needed for the TFT but needed for the SD card reader
#define LoRa_MOSI    27     // GPIO27 -- SX1278's MOSI
#define LoRa_CS      18     // GPIO18 -- SX1278's CS
#define LoRa_RST     23     // GPIO14 -- SX1278's RESET
#define LoRa_DI0     26     // GPIO26 -- SX1278's IRQ(Interrupt Request)
#define Lora_BAND    868E6  // Frequency 868mHz - change to whichever frequency you are using

// Define the variables for the received LoRa packet
String   rssi = "RSSI --"; // Only if you need these
String   packSize = "--";
String   packet;
uint32_t packetSize;
int32_t  packetsReceived = 0;

// Define the GPIO's for the TFT - T-Beam
#define _cs    2  // goes to TFT CS
#define _dc   32  // goes to TFT DC
#define _mosi 13  // goes to TFT MOSI
#define _sclk 14  // goes to TFT SCK/CLK
#define _rst  33  // goes to TFT RESET
#define _miso  4  // may not be necessary to connect to the TFT
//       3.3V     // goes to TFT LED  
//       5v       // goes to TFT Vcc
//       Gnd      // goes to TFT Gnd

// Uses software SPI
Adafruit_ILI9341 tft = Adafruit_ILI9341(_cs, _dc, _mosi, _sclk, _rst);

// Define GPIO's for GPS & baud rate
static const int GPS_RX = 12, GPS_TX = 15; // NOTE - The TX and RX pins are marked incorrectly on the T-Beam !!
static const uint32_t GPSBaud = 9600;

// Define an instance of the TinyGPS++ object
TinyGPSPlus GPS;

// The serial connection to the GPS device
SoftwareSerial ss(GPS_RX, GPS_TX);

.
.
.

setup()
{
  // Start the serial port for debugging 
  Serial.begin(115200);
  while (!Serial)
  {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  tft.begin();
  tft.setRotation(1);
  tft.setTextColor(ILI9341_WHITE,ILI9341_BLACK);
  tft.setTextSize(1);
  tft.fillScreen(ILI9341_BLACK);
  tft.setCursor(0, 0);
  tft.println("TFT Initialized OK...");

  // Initialize SD Card
  pinMode(SS, OUTPUT);

  delay(2000); // Allow some time so as to view the display
  
  if (SD.begin(SD_cs, _mosi, _miso, _sclk))
  {
    tft.println("SD card initialized OK...");

    // If the test.txt file doesn't exist
    // create the file on the SD card and write to it
    myFile = SD.open("test.txt", FILE_WRITE);
  
    // If the file opened okay, write to it:
    if (myFile)
    {
      Serial.print("Writing to test.txt...");
      myFile.println("This is a test");
      // Close the file:
      myFile.close();
      Serial.println("Done.");      
    }
    else
    {
      // if the file didn't open, print an error:
      tft.println("Error opening test.txt for writing");
    }
    
    // Re-open the file for reading:
    myFile = SD.open("test.txt");
    if (myFile)
    {
      Serial.println("test.txt:");
    
      // Read from the file until there's nothing else in it:
      while (myFile.available())
      {
        Serial.write(myFile.read());
      }
      // close the file:
      myFile.close();
    }
    else
    {
      // If the file didn't open, print an error:
      tft.println("Error opening test.txt for reading");
    } 
  }
  else
  {
    tft.println("SD card initialization FAILED...");
  }

  Wire.begin(BMP280_SDA, BMP280_SCL);

  BMP280.begin(); // initialise BMP280 for continuous measurements

  // Start GPS Services
  ss.begin(GPSBaud);
  
  // Start the LoRa service  
  LoRa.setPins(LoRa_CS, LoRa_RST, LoRa_DI0);
   
  if (!LoRa.begin(Lora_BAND))
  {
    tft.println("Starting LoRa failed...");
    while (1); // HALT
  }
  
  tft.println("LoRA initilazed OK...");

  // Put LoRa module into receive mode
  LoRa.receive();
}

Hope this helps getting you up and running...

Enjoy !!

lemmingDev, 2021-03-03:
FYI - I've accessed some pins connected to the LoRa module before by soldering to the pins. It's a bit hacky, but if you're in desperate need for more pins and not using LoRa, then you could try it...

FLYINGSC0T, 2021-03-03:
Hi lemmingDev, Thanks for the info. In fact, I did already think about trying your suggestion to connect to the pins that the LoRa module uses. It might even get the display to work with the TFT_eSPI library by Bodmer. I've got a spare T-Beam so will give it a shot and post back whether it works or not. Re my comment of the 12 June 2019 - did you connect the SPI display SCLK, MOSI and MISO pins to the ones that the LoRa module uses? Thanks for your support.

lemmingDev, 2021-03-03:
@FLYINGSC0T 

I can't remember exactly which pins I used as it was so long ago, but I do remember there was no need to solder to any of the LoRa pins to get it working

lemmingDev, 2021-03-03:
It's also probably not so hard to edit the library to allow for custom SPI pins

ismailox26, 2021-03-24:
> Pin Name	HSPI	VSPI
> GPIO Number
> CS0*	15	5
> SCLK	14	18
> MISO	12	19
> MOSI	13	23
> QUADWP	2	22
> QUADHD	4	21Unfortunately the TTGO PIN mapping not same as ESP32 documentation, i.e: MISO 12 and 19 both not exposed (TTGO uses 19 for MISO but PIN 12 not exposed). Any information if PIN 12 used by GPS?

lah irham lik lwaliddin akhoya

sergimn, 2021-04-02:
I'm having similar issues because I need the GPIO12/MTDI to use the ESP-Prog debuging board. I guess the only workaround is to directly solder the connection, right?

parahooners, 2021-05-31:
having a hard time getting the TFT LCD to work on the T beam..

have belled all cables and connections to check continuity- and tested the screen on another esp32 which work fine.

Here is my set up

----------------------------------------------------------------------------------------------------
#include <SPI.h>
#include <TFT_eSPI.h> // Hardware-specific library
 #include <User_Setups/Setup22_TTGO_T4.h>   
#define TFT_GREY 0x5AEB
#define TFT_MISO 4
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   2  // Chip select control pin
#define TFT_DC    32  // Data Command control pin // double scheck matty
#define TFT_RST   33  // Reset pin (could connect to RST pin)
#define TFT_RST  -1
TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

---------------------------------------------------------------------------------------------------------
User set up file

----------------------------------------------------------------------------------------------------------------
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Define STM32 to invoke optimised processor support (only for STM32)
//#define STM32

// Defining the STM32 board allows the library to optimise the performance
// for UNO compatible "MCUfriend" style shields
//#define NUCLEO_64_TFT
//#define NUCLEO_144_TFT

// STM32 8 bit parallel only:
// If STN32 Port A or B pins 0-7 are used for 8 bit parallel data bus bits 0-7
// then this will improve rendering performance by a factor of ~8x
//#define STM_PORTA_DATA_BUS
//#define STM_PORTB_DATA_BUS

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
//#define TFT_PARALLEL_8_BIT

// Display type -  only define if RPi display
//#define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
#define ILI9341_DRIVER       // Generic driver for common displays
//#define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
[... 271 lines cut ...]
// #define SPI_FREQUENCY  55000000 // STM32 SPI1 only (SPI2 maximum is 27MHz)
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
//#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

parahooners, 2021-05-31:
[image]

stiliajohny, 2021-11-21:
Hi all, 
Is there any update on this . 
I am trying to connect an SD card it doesn't seem to be working

joshbillions, 2023-02-03:
I was having similar issues when trying to interface with Adafruit's MAX31856 breakout via software SPI.

There's no issue with connecting multiple devices in parallel to the SPI bus, so I added leads from the LoRa WAN's IC (HPD13A) to the MAX31856 breakout and everything works as expected.

[image]

On the LoRa WAN IC, starting closest to the USB connector and moving towards the antenna:
MISO / SDO
MOSI / SDI
SCK

Then use any GPIO pin (33 in my case above) to control the CS pin on whatever you're interfacing with.

chinswain, 2024-05-19:
@joshbillions - do you know if these are the same lora module pins on the 1.1 version?