LilyGO/TTGO-T-Beam #11: Schematic for recent T-BEAM

pamribeirox, 2018-10-03:
Hello,
I need the schematic diagram for my T-BEAM (T22 V07 20180711)
Here can I find it?
I've found a few schematics on the net but they seem to be from an older version as some things don't match.

regards.

hasiflo, 2018-10-08:
I am also interested in the actual schematic.
@pamribeirox  perhaps you can post the scematics, which you have found already?

pamribeirox, 2018-10-10:
Nops, the old version I've found is at the end of this page:
http://tinymicros.com/wiki/TTGO_T-Beam

xhalo32, 2019-10-21:
I would love to see the schematics as well. Trying to debug a malfunctioning board is a real pain without them!

Internetwurm, 2019-10-23:
Hello,

maybe i found the right one?

https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkII

a bit down on that page there is the Link "Schematics Updated" that points to this:

https://github.com/Xinyuan-LilyGO/TTGO-T-Beam/raw/master/schematic/T_BeamV1.0.pdf

Hopefully it will be the right one and actual Version.

HouzuoGuo, 2022-06-12:
Here's the schematic of the latest revision of the board on sale as of June 2022: https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/LilyGo_TBeam_V1.1.pdf

gschorcht, 2025-05-12:
I am urgently looking for the schematic of revision v1.0?

What is the reason that only the schematic of the latest revision is available here and older revisions are removed? Older board revisions are still in use. Please provide the schematic for all revisions.