LilyGO/TTGO-T-Beam #12: T-Beam Program Download Erratic

chillmf, 2018-10-19:
I have several t-beam boards and have successfully downloaded applications such as SoftRf and ESP-sc-gway.

these applications run well and I am pleased with the operation.

However, I have continued to have problems downloaded compiled code using both the Arduino IDE and ESP Download Tool V3.6.4.

Usually I get the error message "A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header"

Has anyone else had a similar problem.

If so have you found a solution.

Thanks for the help.

Charley

lemmingDev, 2018-10-21:
I would try a better USB cable

chillmf, 2018-10-23:
Thanks for the suggestion, it has worked!

Regards,

charley

 

Links:
------
[1]
https://github.com/LilyGO/TTGO-T-Beam/issues/12#issuecomment-431703422
[2]
https://github.com/