LilyGO/TTGO-T-Beam #17: No I2C detected on board V07

happygaru, 2019-01-22:
Hi,

I have a T-Beam version T22_V07 (20180711). 
GPS and LoRa work great, but I can't connect to a barometric sensor on either I2C bus -  the device is not found on 1st I2C bus (2/13) nor 2nd I2C bus (21/22) work.

I have followed the advice posted on https://github.com/lyusupov/SoftRF/issues/32, but no luck.
The sensor and code work with an Arduino Uno. The code is here:
https://github.com/bluerobotics/BlueRobotics_MS5837_Library/tree/master/examples/MS5837_Example

Any ideas how to get I2C to work on V07?

Cheers!

proffalken, 2019-03-01:
@happygaru can you put the code on here to show where you placed the "wire begin"? I'm having issues getting the basic example working from the Adafruit library...

happygaru, 2019-03-01:
Sure, here's the adapted I2C scanner from Nick which I found really helpful:

`// I2C Tester
// Written by Nick Gammon
// Date: 20th April 2011

#include <Wire.h>

void setup() {
  Serial.begin (115200);

  // Wait for serial port to connect
  while (!Serial) 
    {
    }

  Serial.println ();
  Serial.println ("I2C scanner. Scanning ...");
  byte count = 0;

  Wire.begin(21,22);  // for T-Beam pass SDA and SCL GPIO pins
  for (byte i = 8; i < 120; i++)
  {
    Wire.beginTransmission (i);
    if (Wire.endTransmission () == 0)
      {
      Serial.print ("Found address: ");
      Serial.print (i, DEC);
      Serial.print (" (0x");
      Serial.print (i, HEX);
      Serial.println (")");
      count++;
      delay (1);  // maybe unneeded?
      } // end of good response
  } // end of for loop
  Serial.println ("Done.");
  Serial.print ("Found ");
  Serial.print (count, DEC);
  Serial.println (" device(s).");
}  // end of setup

void loop() {}
`

mgpiccinini, 2021-09-16:
Hi,
I am a beginner and have a TTGO T-Beam device with wifi, BT, GPS and LoRa.
I am trying to run a simple program to start to understand how it works and I am stuck at the AXP192 initialization process. It seems that the AXP192 initialization fail and I don´t know reasons for this behavior.

I used the code above and the result is bellow and it seems again that nothing is happening when I am trying to initialize the I2C communication.

Please can someone can help me with some idea?

21:59:40.869 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:59:40.869 -> configsip: 0, SPIWP:0xee
21:59:40.869 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:59:40.869 -> mode:DIO, clock div:1
21:59:40.869 -> load:0x3fff0018,len:4
21:59:40.869 -> load:0x3fff001c,len:1044
21:59:40.869 -> load:0x40078000,len:10124
21:59:40.869 -> load:0x40080400,len:5856
21:59:40.869 -> entry 0x400806a8
21:59:41.009 -> 
21:59:41.009 -> I2C scanner. Scanning ...
21:59:41.009 -> Done.
21:59:41.009 -> Found 0 device(s).

Thank you in advance.