LilyGO/TTGO-T-Beam #19: GPS Sleep/Off

oricoh, 2019-03-18:
How do I put the GPS into sleep mode or turn it off? It draws a lot of power when the device is in sleep mode.

zacharydrew, 2019-03-21:
You can turn it off via the serial interface. Take a look at the [u-blox 6
Receiver Description](http://tinymicros.com/mediawiki/images/3/3f/ublox_NEO-6_Protocol_Specification.pdf). Section 9.5 describes turning it on and off. Section 9.3 describes power saving mode.

oricoh, 2019-03-22:
Thanks @zacharydrew I did found that out eventually. Nevertheless the GPS should have been assigned a GPIO by design.

zacharydrew, 2019-03-22:
Agreed, but since you can do it without reworking the board the board the fix is cheap.

I will have to jumper GPS PPS to an ESP32 GPIO to get a more accurate clock when my boards arrive in a few weeks.

I ordered what looks like a newer board revision with a uBlox M8N GPS and uFL antenna connector. They could have installed those components without reving the board but I am crossing my fingers. I ordered the board specifically for the GPS and TF card rather than Lora hardware.

I would love if they

1) posted the newer schematic and BOM
2) connect GPS PPM to ESP32 GPIO
3) released a lower cost board with everything but the Lora components*

*I am really surprised I can’t find a an esp board with GPS and TF anywhere

oricoh, 2019-03-22:
1. I need the Lora....
2. After shutting down the GPS via serial, I still read 20mAh in deep sleep. I need to bring down to under 1mA to the uA range to save battery. I am still trying to figure out what's using 20mAh... any ideas?
3. Can you post a link to the new board?

zacharydrew, 2019-03-22:
To each their own.

Any reason you believe it’s GPS and not other components responsible for that power draw?

https://m.aliexpress.com/item/32914511463.html?trace=wwwdetail2mobilesitedetail&productId=32914511463&productSubject=TTGO-T-Beam-433MHZ-868MHZ-915MHZ-ESP32-WiFi-wireless-Bluetooth-Module-ESP-32-GPS-NEO-M8N

lemmingDev, 2019-03-28:
This link seems to have it covered - if your GPS module is genuine

https://github.com/JoepSchyns/Low_power_TTGO_T-beam

SimonRafferty, 2019-11-02:
I suspect most (mine anyway) of the GPS modules are fake.  Mine does not respond to any command strings - but works OK as a GPS.  If you try to put it to sleep - it just carries on the same.
What's the answer?

The board has an AXP192 power controller to allow the ESP to switch peripherals on & off.
Have a look at the AXP202X Library by Lewis He:
https://platformio.org/lib/show/6657/AXP202X_Library

Add the following to your sketch:
`
#include <axp20x.h>
AXP20X_Class axp;
`
And in Setup():
`     
  //Turn everything on
  axp.setPowerOutPut(AXP192_LDO2, AXP202_ON);
  axp.setPowerOutPut(AXP192_LDO3, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC1, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC2, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC3, AXP202_ON);
  axp.setPowerOutPut(AXP192_EXTEN, AXP202_ON);
  axp.setDCDC1Voltage(3300); //Set Pin header 3.3V line to 3.3V.  Processor is happy down to 1.8V which saves power

  //I don't think this board has a built in LED, other than the Charge LED, controlled by this:
  //  AXP20X_LED_OFF,
  //  AXP20X_LED_BLINK_1HZ,
  //  AXP20X_LED_BLINK_4HZ,
  //  AXP20X_LED_LOW_LEVEL,

  axp.setChgLEDMode(AXP20X_LED_OFF);

`
The above powers everything up.  If you replace AXP202_ON with AXP202_OFF, it will turn the peripherals off.  You'll have to experiment to see which peripheral is the GPS.

As in the example above, in addition to switching on & off, you can change the voltage.  this table (from AXP202X.h) shows the usable voltage range for the different peripherals.  You can save a lot of power just by trimming down the voltages.

>     !! Chip resource table
>     | CHIP     | AXP173           | AXP192           | AXP202           |
>     | -------- | ---------------- | ---------------- | ---------------- |
>     | DC1      | 0v7~3v5  /1200mA | 0v7~3v5  /1200mA | X                |
>     | DC2      | 0v7~2v275/1600mA | 0v7~2v275/1600mA | 0v7~2v275/1600mA |
>     | DC3      | X                | 0v7~3v5  /700mA  | 0v7~3v5  /1200mA |
>     | LDO1     | 3v3      /30mA   | 3v3      /30mA   | 3v3      /30mA   |
>     | LDO2     | 1v8~3v3  /200mA  | 1v8~3v3  /200mA  | 1v8~3v3  /200mA  |
>     | LDO3     | 1v8~3v3  /200mA  | 1v8~3v3  /200mA  | 0v7~3v3  /200mA  |
>     | LDO4     | 0v7~3v5  /500mA  | X                | 1v8~3v3  /200mA  |
>     | LDO5/IO0 | X                | 1v8~3v3  /50mA   | 1v8~3v3  /50mA   |

fphammerle, 2021-05-22:
On T-Beam v1.1 the GPS module is connected to AXP192's `LDO3` pin:

```cpp
// https://github.com/lewisxhe/AXP202X_Library
#include <axp20x.h>
// ...
if(axp.begin(Wire, AXP192_SLAVE_ADDRESS) == AXP_FAIL) {
  Serial.println(F("failed to initialize communication with AXP192"));
}
// ...
if(axp.setPowerOutPut(AXP192_LDO3, AXP202_OFF) == AXP_PASS) {
  Serial.println(F("turned off GPS module"));
} else {
  Serial.println(F("failed to turn off GPS module"));
}
```