LilyGO/TTGO-T-Beam #20: Repalcement of 18650 Battery

hopexxx, 2019-04-06:
it is possible to replace the 18650 battery with lipo battery?

lemmingDev, 2019-04-14:
Will be fine if it's 3.7V

SimonRafferty, 2019-11-02:
I removed the battery holder & soldered a 1S 1000mAH battery to the pads.  The original battery was too chunky for my application & the new one is much smaller.  It works just fine.

Prophet15, 2022-03-16:
Anda how did you do It? I'm just starting with this... And I was just wondering if you would have a dummies guide for a new battery installation.

Would you share a link to buy the 1000mAh batteries?

What are the batterie's specifications needed to change them? (Voltage, etc) 

Thanks!!

SimonRafferty, 2022-03-16:
Do you have a soldering iron?
Use it to remove the original battery holder.  In fact, you could just leave it in place if it isn't in the way.
Buy a replacement battery - I can't remember where mine came from, probably Amazon.  Any 3.7V LiPo cell will do.
Cut the connector off the battery (if it has a connector).
Solder the Red battery wire to the same end as the + terminal on the old battery holder
Solder the Black wire to -
It's no more complicated than that.

Prophet15, 2022-03-16:
Yes, I do.  

Ok thanks, I'm gonna try It.

I will look for the batteries on Amazon for sure.

What about replace the GPS antenna? Is it worth it? How do you do it?