LilyGO/TTGO-T-Beam #22: Dead Board?

krab-skunk, 2019-06-16:
Hi, i have 2 ttgo t-beam lora32 868Mhz boards.

They worked fine for 5 months or so, but since today
one board, stop working.
It's not:
 - recognized by computer (using platform.io)
 - the LED 1014 is not turning on anymore once the switch is set to ON.
 
All i got is, once powered on USB, i got a green led as per the picture:
https://imgur.com/a/IgGZHNV
I tried resetting it, nothing happened.

Is it dead or you would recommend me some things to try before throwing it, i have a multimeter but no schematic.

Thanks a lot

SimonRafferty, 2019-11-02:
I had the same problem out of the box!  The 3.3V pin header read 1.8V and nothing else was powered at all.  The reason is that it has an AXP192 power controller to allow the ESP to switch peripherals on & off.
Have a look at the AXP202X Library by Lewis He:
https://platformio.org/lib/show/6657/AXP202X_Library

Add the following to your sketch:
`
#include <axp20x.h>
AXP20X_Class axp;
`
And in Setup():
`     
 /*
    !! Chip resource table
    | CHIP     | AXP173           | AXP192           | AXP202           |
    | -------- | ---------------- | ---------------- | ---------------- |
    | DC1      | 0v7~3v5  /1200mA | 0v7~3v5  /1200mA | X                |
    | DC2      | 0v7~2v275/1600mA | 0v7~2v275/1600mA | 0v7~2v275/1600mA |
    | DC3      | X                | 0v7~3v5  /700mA  | 0v7~3v5  /1200mA |
    | LDO1     | 3v3      /30mA   | 3v3      /30mA   | 3v3      /30mA   |
    | LDO2     | 1v8~3v3  /200mA  | 1v8~3v3  /200mA  | 1v8~3v3  /200mA  |
    | LDO3     | 1v8~3v3  /200mA  | 1v8~3v3  /200mA  | 0v7~3v3  /200mA  |
    | LDO4     | 0v7~3v5  /500mA  | X                | 1v8~3v3  /200mA  |
    | LDO5/IO0 | X                | 1v8~3v3  /50mA   | 1v8~3v3  /50mA   |
    */
  //Turn everything on
  axp.setPowerOutPut(AXP192_LDO2, AXP202_ON);
  axp.setPowerOutPut(AXP192_LDO3, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC1, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC2, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC3, AXP202_ON);
  axp.setPowerOutPut(AXP192_EXTEN, AXP202_ON);
  axp.setDCDC1Voltage(3300); //Set Pin header 3.3V line to 3.3V.  Processor is happy down to 1.8V which saves power

  //I don't think this board has a built in LED, other than the Charge LED, controlled by this:
  //  AXP20X_LED_OFF,
  //  AXP20X_LED_BLINK_1HZ,
  //  AXP20X_LED_BLINK_4HZ,
  //  AXP20X_LED_LOW_LEVEL,

  axp.setChgLEDMode(AXP20X_LED_OFF);

`
Everything will come back to life!

mattbk, 2023-08-07:
In case it helps, one of the lines above looks like it should be split into two:
```
#include <axp20x.h> 
AXP20X_Class axp;
```