LilyGO/TTGO-T-Beam #24: How to change modulation to communicate with Heltec?

cnorman27, 2019-07-26:
I have the Heltec ESP32+OLED+LoRA module.  I can tell by using SDR# that the modulation is different but came find support for either one to make them communicate.

anthonyvage, 2020-01-29:
Do you find a solution for your problem? I have the same problem.

closed 2020-01-29 by cnorman27

IZ0DZX, 2020-01-29:
> Closed  #24 .

Hello
I have not jet resolved the communication problem between Heltec board 1276x and TTGO T Beam. This is ok between the two Heltecs i own and also between the two T-Beams Where is the problem? please help
Thanks
Inviato da Libero Mail per Android mercoledì,

IZ0DZX, 2020-01-29:
Hello
I have not jet resolved the communication problem between Heltec board 1276x and TTGO T Beam. This is ok between the two Heltecs i own and also between the two T-Beams Where is the problem
--
Inviato da Libero Mail per Android mercoledì,

anthonyvage, 2020-01-30:
I resolved the problem. Check the settings for spreading factor. It seems that heltec lora library has different default settings as the library ttgo uses.