LilyGO/TTGO-T-Beam #27: Range?

vatsalkansara2, 2019-08-15:
I have two TTGO boards one is TTGO - T Beam and TTGO OLED V2 
I have connected two boards with sample LoRa sender receiver which works fine.

But after like 150m it starts to drops packet.

Can anyone tell me is it the maxmimum range of it or i am doing something wrong?
What should i have to do for improving range?

Thanks

netmonk, 2019-09-07:
i succeeded around 280m without drop https://www.instagram.com/p/B2HfbZzHpo_

wolfpcgn, 2019-09-28:
The range strongly depends on the frequency and the line of sight. If you use the T-Beam on the ground in a city you will get only a few 100m. If you are in the air the range can be several 10km. 
As a rule of thumb: the higher the frequency the more important becomes the line of sight. There is a good App from R&S (Rhode und Schwarz) for the estimation of the distance depending on power, sensitivity of receiver, and height above ground.

vatsalkansara2, 2019-10-02:
i have gone about 950m (Approx~1km) in straight LOS

closed 2019-10-02 by vatsalkansara2

YD0NXX, 2021-02-28:
Using the 433 MHz board, with SF12 and power at 20, I get 4 KM in LOS (from my vehicle that was on a freeway overpass to my hamradio antenna on the tower at home). That packet was with SNR of -20.

pimalu, 2023-05-06:
Looks like it has very narrow use case for outdoors-hiking, unless you are on Everest summit....