LilyGO/TTGO-T-Beam #30: Pinout diagram corrections

lemmingDev, 2019-09-07:
Hi

The pinout diagram should be updated to reflect that pins 32 and 33 are only capable of output (GPO), and pins 34, 35, 36 and 39 are only capable of input (GPI).

See this official Espressif document here:
[http://esp32.net/images/Espressif/ESP-WROOM-03/Espressif_ESP-WROOM-03_Pinout.pdf](http://esp32.net/images/Espressif/ESP-WROOM-03/Espressif_ESP-WROOM-03_Pinout.pdf)

Also, touch inputs T1 (pin 0 - USB serial chip) and T6 (pin 14 - LED) have resistors connected to them and won't work for touch, so those labels should be removed.

Skipper-is, 2020-11-07:
Are any of the pins set up for reading battery level?

lemmingDev, 2020-11-09:
Pin 35 I think

So something like:

float vBat;
vBat = analogRead(35) * 2.0 * (3.3 / 1024.0);
Serial.print("Battery voltage: ");
Serial.print(vBat);
Serial.println("V");

There is also a more precise way of using a factory set value
Will check how it's done again and let you know

lemmingDev, 2020-11-09:
This sketch for the TTGO T-Display, among other things, shows how to use the e-fuse ESP_ADC_CAL_VAL_EFUSE_VREF value to get the accurate battery reading 

https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/TFT_eSPI/examples/FactoryTest/FactoryTest.ino

Let me know if you have trouble deciphering it to use

lemmingDev, 2020-11-09:
This sketch sets up a Classic BT serial port that you can monitor for a voltage readout.

You'll see that it goes up when the USB is plugged in, and down when the USB is removed

'''
#include "BluetoothSerial.h"
#include "esp_adc_cal.h"

#define ADC_EN              14  //ADC_EN is the ADC detection enable port
#define ADC_PIN             35  //ADC_PIN is the ADC pin the battery is connected to through a voltage divider

int vref = 1100;                //Default value. We'll get a more accurate value from the efuses

BluetoothSerial SerialBT;

void showVoltage()
{
    static uint64_t timeStamp = 0;
    if (millis() - timeStamp > 1000) {
        timeStamp = millis();
        uint16_t v = analogRead(ADC_PIN);
        float battery_voltage = ((float)v / 4095.0) * 2.0 * 3.3 * (vref / 1000.0);
        String voltage = "Voltage :" + String(battery_voltage) + "V\n";
        SerialBT.println(voltage);
    }
}

void setup()
{
    SerialBT.begin("ESP32test"); //Bluetooth device name
    /*
    ADC_EN is the ADC detection enable port
    If the USB port is used for power supply, it is turned on by default.
    If it is powered by battery, it needs to be set to high level
    */
    pinMode(ADC_EN, OUTPUT);
    digitalWrite(ADC_EN, HIGH);

    esp_adc_cal_characteristics_t adc_chars;
    esp_adc_cal_value_t val_type = esp_adc_cal_characterize((adc_unit_t)ADC_UNIT_1, (adc_atten_t)ADC1_CHANNEL_6, (adc_bits_width_t)ADC_WIDTH_BIT_12, 1100, &adc_chars);
    //Check type of calibration value used to characterize ADC
    if (val_type == ESP_ADC_CAL_VAL_EFUSE_VREF) {
        SerialBT.printf("eFuse Vref:%u mV", adc_chars.vref);
        vref = adc_chars.vref;
    } else if (val_type == ESP_ADC_CAL_VAL_EFUSE_TP) {
        SerialBT.printf("Two Point --> coeff_a:%umV coeff_b:%umV\n", adc_chars.coeff_a, adc_chars.coeff_b);
    } else {
        SerialBT.println("Default Vref: 1100mV");
    }
    SerialBT.println();
}

void loop()
{
  showVoltage();
}
'''

BrugerX, 2023-07-11:
> Hi
> 
> The pinout diagram should be updated to reflect that pins 32 and 33 are only capable of output (GPO), and pins 34, 35, 36 and 39 are only capable of input (GPI).
> 
> See this official Espressif document here: http://esp32.net/images/Espressif/ESP-WROOM-03/Espressif_ESP-WROOM-03_Pinout.pdf
> 
> Also, touch inputs T1 (pin 0 - USB serial chip) and T6 (pin 14 - LED) have resistors connected to them on the T-Beam and won't work for touch, so those labels should be removed.
> 
> Here is a file you could use [image]

Hello Lemming,

I am trying to use the T-Beam to read a soil moisture sensor. I have used a voltage divider circuit with R1 being 10k and R2 being 20k. I get a reading from the output of the sensor of approx. 3.6V and approx. 2.4V from the output of the voltage divider.

However, both pin 33 and pin 35 read 0 value using the Arduino analogRead with the 2.4V. Using the 3.6V directly I do get a result, however they're very different, with pin 33 reading approx. 1100 and pin 35 reading approx. 2800.

Do you have any idea, why this is happening?