LilyGO/TTGO-T-Beam #32: Version T22 V1.0: LED Built-in GPIO

rcomellas, 2019-09-14:
I have tested the blink sketch from the examples folder in the new version T22 V1.0 but it doesn't work with GPIO14.
Anyone has the right IO?

SimonRafferty, 2019-11-02:
I do not believe the board has a built-in LED other than the charging LED.  However, you can control this!

Have a look at the AXP202X Library by Lewis He:
https://platformio.org/lib/show/6657/AXP202X_Library

Add the following to your sketch:
`
#include <axp20x.h>
AXP20X_Class axp;
`
And in Setup():
`     
  //Turn everything on
  axp.setPowerOutPut(AXP192_LDO2, AXP202_ON);
  axp.setPowerOutPut(AXP192_LDO3, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC1, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC2, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC3, AXP202_ON);
  axp.setPowerOutPut(AXP192_EXTEN, AXP202_ON);
  axp.setDCDC1Voltage(3300); //Set Pin header 3.3V line to 3.3V.  Processor is happy down to 1.8V which saves power

  //I don't think this board has a built in LED, other than the Charge LED, controlled by this:
  //  AXP20X_LED_OFF,
  //  AXP20X_LED_BLINK_1HZ,
  //  AXP20X_LED_BLINK_4HZ,
  //  AXP20X_LED_LOW_LEVEL,

  axp.setChgLEDMode(AXP20X_LED_OFF);

`

luckynrslevin, 2020-06-07:
Following code wokrs for me:
https://github.com/luckynrslevin/TTGO-T-Beam-Blinking