LilyGO/TTGO-T-Beam #33: Ports are displayed in wrong order on V1.0

Internetwurm, 2019-10-22:
Hello LilyGo,

this is a really nice product.
the pinout picture shows Port 21 and 22 in right order.
Picture and module writing is the same.

https://github.com/kizniche/ttgo-tbeam-ttn-tracker/blob/master/img/T-BEAM-Rev1-01.jpg

I am not really sure if this picture is done by LilyGo.
But the style looks similar to the picture for the earlier version. 

On this picture the module ordering 21 and 22 is the same but the writing right next to it display it in different order. 22 and 21 as also SDA and SCL are changed to the earlier version and pictures.
It may make a complictaed difference if somone doesn´t check more then twice and just relay to the picture. 

Thanks for this nice and good stuffed product.

regards
[image]

lyusupov, 2019-11-08:
Good catch!

I would suggest you to send your request to LilyGO **sales** staff.
( They create marketing info. Engineers don't do that. )
They typically fix the pictures within few days upon receive of customer's request.

Use communication method of the platform you've purchased the board onto (AliExpress, E-bay, Amazon, Banggood, …)

Internetwurm, 2019-11-08:
Yes, i think it was obvious to post it here.
I will send the info to sales directly.

Thanks for reminding. ;-)

closed 2019-11-08 by Internetwurm