LilyGO/TTGO-T-Beam #36: Can't get GPS data

guilhermesdas, 2020-02-18:
Hi, I have a TTGO T22_V07 board and it's not receiving gps data. I'm running this [code](https://github.com/LilyGO/TTGO-T-Beam/blob/master/GPS/GPS.ino) in Arduino IDE. I already tried to upload with these boards: TTGO Lora32-OLED V1, TTGO T1 and ESP32 Dev Module.

The GPS blue red is always turn on but the red led never flashes. I already tried to test in the window outside my house but nothing changes.

japanoronha, 2020-04-19:
I have the same problem, but my board is a T-Beam. I run the GPS code in the correct board and the output is: 

No GPS data received: check wiring
Latitude  : 0.00000
Longitude : 0.0000
Satellites: 0
Altitude  : 0.00M
Time      : 0:0:0

minusplusminus, 2020-06-04:
same problem

gagliardi67, 2020-06-10:
TX IO34 RX IO12 : ESP 32 input

cvanaalst, 2020-07-02:
I have the same problem (T-beam v1 with Neo-M8N GPS), I cannot get the gps to work - I have bought 2  - lora works, but I have tried everything - the GPS does not work on either of the two ...
**********************
No GPS data received: check wiring
Latitude  : 0.00000
Longitude : 0.0000
Satellites: 0
Altitude  : 0.00M
Time      : 0:0:0
Speed     : 0.00
**********************

In addition I tried the TinyGPS++ lib and this shows me that nothing is being read over the serial connection from the Neo-M8N GPS ....

minusplusminus, 2020-07-03:
Figured out that the AXP192 has to be enabled. If not, gps won't be active. See this repo for more info: https://github.com/kizniche/ttgo-tbeam-ttn-tracker

rlpdavid, 2020-07-25:
Did anyone ever make this work? Looks like the AXP PMU itself is the culprit

Sherlock-Homie, 2022-01-06:
I have been using the ESP TTGO T-Beam V1.1 and found the examples from LilyGo, GlacierJay often initialize the GPS with different pin mappings to what my device requires (probably using a different version or flavour from LilyGO)

Ensure you are using Pin 34 for TX, and Pin 12 for RX.

your code should look something like - Serial1.begin(9600, SERIAL_8N1, 34, 12); 

I just modified the example from GlacierJay and managed to get it to work: https://github.com/glacierjay/T_beam/tree/master/gps/tbeam_gps