LilyGO/TTGO-T-Beam #38: How to measure battery level?

andrecurvello, 2020-06-03:
Hello,

There is any example on how to measure the battery level of hte TTGO T-Beam?

Thank you.

KenADev, 2020-07-12:
With v1.0+ you can query the AXP chip with:
`axp.getBattVoltage()/1000.0`

fphammerle, 2021-05-14:
tested on t-beam v1.1:

```c
#include <Wire.h>
// https://github.com/lewisxhe/AXP202X_Library
#include <axp20x.h>

AXP20X_Class axp;

void setup() {
  // ...
  Wire.begin(21, 22);
  if(axp.begin(Wire, AXP192_SLAVE_ADDRESS) == AXP_FAIL) {
    Serial.println(F("failed to initialize communication with AXP192"));
  }
}

void loop() {
  // ...
  Serial.print(axp.getBattVoltage());
  Serial.println(" mV");
}
```

jgamage91, 2023-10-28:
Hi, is the library https://github.com/lewisxhe/AXP202X_Library no longer maintained? Would there be an alternate lib?

chinswain, 2024-05-14:
Lewis has created a library for all of the power chips LilyGO use:

https://github.com/lewisxhe/XPowersLib/blob/master/examples/AXP202_Example/AXP202_Example.ino