LilyGO/TTGO-T-Beam #42: Uploading sketch issue

pbravoz2, 2020-10-08:
Hi there,
               uploading the sketch to T-Beam in Arduino IDE 1.8.12 throws me the classic ESP32 error: **Failed to connect to ESP32: Timed out waiting for packet header**. The classic solution for the other ESP32 boards is to place a 10uF capacitor between the EN PIN and GND. But T-Beam doesn't have the EN pin! How can I solve it?
I am very thankfull for your help.
Greetings, Pedro from Chile.

dnwk, 2021-03-08:
I have the same issue

tezmo, 2021-03-23:
Same issue here..

tezmo, 2021-03-29:
Actually, my issue was the driver. Checked on another PC, worked. Uninstalled and reinstalled driver, rebooted PC. issue resolved.

GeorgeBR, 2022-03-11:
Same issue for me - neither Arduino IDE (v1.8.15) and PlatformIO (in Visual Studio Code). Board shows up on USB. But when trying to upload no response from board - nothing!
So far, hugely unimpressed with this board - wish I'd bought the individual components