LilyGO/TTGO-T-Beam #47: What is the LoRaWAN version (MAC) supported by TTGO T-Beam?

HouzuoGuo, 2021-01-28:
Hello!

I'm making a migration from TheThingsNetwork v2 to their new v3 infrastructure. There's a new dropdown selection when registering a new device:
[image]

According to the TTN reference (https://www.thethingsindustries.com/docs/reference/glossary/#lorawan-version) the version selection is:

"The LoRaWAN version is the LoRa Alliance LoRaWAN specification your device conforms to, which defines which Media Access Control features it supports."

What's that version currently supported by T-Beam board revision 0.7, board revision V1.0, and board revision V1.1?

mloeschner, 2021-02-12:
The board provides a modem to use LoRa. The LoRaWAN protocol stack is provided by an library (e.g. LMIC, which supports 1.02/1.03). You will find the information there.

HouzuoGuo, 2021-02-12:
@mloeschner Are you sure about that? The doc published by TheThingsIndustries clearly states:

"The Media Access Control version for your device should be provided by the manufacturer in a datasheet"

mloeschner, 2021-02-12:
The statement is correct as one usually gets a device from a commercial manufacturer.