LilyGO/TTGO-T-Beam #51: Poor GPS performance with Rev1.1 20210222 boards

kd2eat, 2021-10-04:
Hi,

I own a few Rev 1.1 20191212 T-Beams.  The GPS works great, getting a lock with 6-9 satellites, even from inside my house.  They routinely got 10-12 satellites if I set it out in my yard.  I recently purchased a few more.  They are Rev 1.1 20210222 boards.  The GPS performance is markedly poorer.  They rarely if ever get a lock from within my home, and even out in the open in my front yard, they only get about 5 satellites.  That takes a long time.

The newer boards shipped with some wonky code that disabled NMEA, with those settings saved to non-volatile RAM.  I dug up the commands and turned on NMEA again.  That seems OK, but perhaps they saved some other whacky receiver settings that negatively impact things?  I've spent a bunch of hours on this, and I'm not really getting anywhere.  At this point, I'm reluctant to buy more T-Beams, as it's beginning to feel like a hardware problem.

Has anyone else seen poor performance of the GPS on the Rev 1.1 20210222 boards?

Mike

daveake, 2021-10-04:
I've also got 2 of the 20210222 boards which I agree have noticeably worse GPS performance than do the 20191212 boards.

I'm using an external amplified GPS antenna.  The older boards quickly get 12 sats with this antenna; the newer boards struggle to get 9 sats.  That's with the same antenna, in the same location, with the same IPX-SMA pigtail, tested minutes apart.

As a test I wrote a small program to echo the NMEA to USB, so I could connect the board to a PC running U-Center.  The results are attached, showing many more visible satellites on the earlier board.

I should note that my 2021 boards also had some firmware that disabled NMEA, saving that config in nvram on the board.  To clear the settings I removed the nvram and shorted the GPS backup battery to be sure.

Dav
[image]
[image]
e

shellinspector, 2021-12-29:
Same issues. Only had a single GPS coordinates fetch on one of the two Rev1.1 20210222 I bought.  New to T-Beam boards.  Just would like to know if the problem is in wiring. I tried running GPS examples from https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series, and the results are poor with TinyGPS++ examples not even detecting a GPS module, asking me to check wiring. Guys, do you have an advice on which program to used just to check the barebones GPS basics to see if my GPS module is alive? Any input would be gold.

> I've also got 2 of the 20210222 boards which I agree have noticeably worse GPS performance than do the 20191212 boards.
>
> I'm using an external amplified GPS antenna. The older boards quickly get 12 [...]

Could you please share your trick/program so I could see if I also detect any GPS sattelites?

> Antenna is total trash. Just replace with bigger one.

Has replacing a battery worked for you? If yes, what program/firmware are you using to check? Which antenna have you bought? Thanks in advance.

mauricecyril, 2022-01-22:
Hi I'm experiencing the same issue with Rev1.1 20210222 boards.

shellinspector, 2022-01-23:
It's not hopeless, guys. The GPS on  Rev1.1 20210222 do work, which is good. They work poorly, which is bad. When taking the board outside with the latest meshtastic firmware 1....49 (or something) and giving it 5-10min, it finds 5-7 satellites. Ordered me a bigger GPS antenna.

mauricecyril, 2022-01-24:
@shellinspector  you're correct. I switched out the stock GPS antenna that came with this revision and then started picking up satellites with the red light on the GPS module blinking.

mauricecyril, 2022-01-24:
Another update, I plugged the stock antenna back in and the satellites pick up. So somehow using the bigger antenna the device was able detect and save the satellites.  Was getting 9 satellites indoor with the bigger antenna and now 5 with the stock one.

HouzuoGuo, 2022-03-11:
Yep - I've got 2 of the boards with a revision number from 2019, and a couple more with the revision number from 2021.

The older boards obtain a GPS location fix within a minute or two using the ceramic patch antenna that came with unit.

The newer boards struggle to get a GPS location fix and it takes 5-6 minutes!

GuillermoHazebrouck, 2022-06-23:
I just opened a new issue but actually it seems related to this one. My T-Beam is not getting any fix at all. Could it be due to a defect antenna? Is there any way to get more detailed information about the status of the board than the info thrown by the http page?

mauricecyril, 2022-06-24:
The only way I was able to get a GPS lock to remain on tbeams  (Rev1.1 20210222 boards) was to change the GPS antenna to a larger ceramic version.

GuillermoHazebrouck, 2022-06-27:
I'll try a new antenna then.
I don't see the exact specifications for the connector, I guess it is an Ipex connector, but are they all the same?

lyusupov, 2022-08-16:
Read this: https://github.com/LilyGO/TTGO-T-Beam/issues/55#issuecomment-1216238743