LilyGO/TTGO-T-Beam #55: No GNSS fix

GuillermoHazebrouck, 2022-06-23:
I can power on the board (via USB cable), connect to the hotspot and access the status monitoring page. However not fix is reported (number of satellites stays at 0), even after several minutes outside.
Board info:
Device Id = 46B8AC
Software version = 1.0-rc9 ESP32
GNSS = U8
Radio = SX1276

lyusupov, 2022-08-16:
It was reported that certain T-Beam batches could have been manufactured with wrong R38 value.

[image]

daveake, 2022-08-16:
That makes sense - when I compared my devices some had a different R38 value.  I don't recall what value but I'll check.

daveake, 2022-08-16:
> By replacing the original antenna I can obtain a fix, but only after some
> minutes. I have not tested it outside, but seems to be taking long anyway.
> Questions: How can I safely measure the resistance, and if the value is
> wrong, can it be replaced?

You can measure it with any multimeter.  Make sure the antenna is
disconnected so it doesn't affect the reading.

It can be replaced.  I don't do much SMD soldering but it's not hard.

TheSkorm, 2025-03-26:
Are you actually sure you were measuring R38. If its the large resistor looking thing near the uFL connector, that's actually R37 inductor. The smaller resistor next to it R38. I don't think the value of this matters too much as its supplying bias voltage to the LNA in the active antenna. Most designs seem to use 22 ohm however I didn't see any improvement with that value.

lyusupov, 2025-03-26:
22 Ohm , 100 Ohm is Ok for R38

There was a batch of T-Beams with something like **10K** value of R38 actually soldered. This happened once in 2021-2022 time frame.
Built-in LNA of supplied GNSS antenna was underpowered and this caused numerous reports of "bad antenna quality" at that  time... 
That was an issue of 1 or 2 batches only. Someone of LilyGO factory personnel had put a wrong reel of resistors into the soldering machine...